ktuukkan/marine-api #25: Provide own DataReader to SentenceReader, Logging

GPSHansl, 2015-01-08:
It is pretty straight forward to have an SentenceReader initialized with either DatagramSocket or an InputStream.
On instantiation with an InputStream, the "DefaultDataReader" tries to read lines. The "available" on the InputStream is not checked in "DefaultDataReader", so there is the posibility (when accessing a SerialPort) that there is no data available, which leads to "WARNING: InputStream read failed
java.io.IOException: Underlying input stream returned zero bytes". 
The next problem here is, that the "SentenceReader.handleException" always logs a warning, even when another "ExceptionListener" is set.

Steps to be taken:
1. most simple: provide a "SentenceReader(DataReader source)" constructor to create the SentenceReader instance

	/**
	 * Creates a SentenceReader.
	 *
	 * @param source DataReaer from which to read NMEA data
	 */
	public SentenceReader(DataReader source) {
		reader = source;
	}

2. simple though: only log an Exception, when no other ExceptionListener is registered:

	void handleException(String msg, Exception ex) {
		if(exceptionListener != null) {
			try {
				exceptionListener.onException(ex);
			} catch (Exception e) {
				LOGGER.log(Level.WARNING, "Exception thrown by ExceptionListener", e);
			}
		} else {
			LOGGER.log(Level.WARNING, msg, ex);
		}
	}

ktuukkan, 2015-01-08:
Added ready() check before reading the input stream and changed logging only if no ExceptionHandler is present, check the snapshots for latest build.

goyalnaresh, 2018-07-09:
ready() check is no longer in branch v0.10.0. 
@ktuukkan 
Is there any (plan to add support of ) custom data reader as suggested in issue description

- provide a "SentenceReader(DataReader source)" constructor to create the SentenceReader instance

    `/**
     * Creates a SentenceReader.
     *
     * @param source DataReaer from which to read NMEA data
     */
    public SentenceReader(DataReader source) {
        reader = source;
    } `

ktuukkan, 2018-07-11:
Hi @goyalnaresh, `ready()` must have been removed while readers were refactored. Might be a problem again, thanks for pointing out. There hasn't been other requests for the constructor after this issue was closed, so haven't thought about it much. But of course it could be done, can you tell some more about your use case and how the constructor would help resolving it?

GPSHansl, 2018-07-12:
Hi @ktuukkan, @goyalnaresh 
I intercept here.

It was my first intention providing a custom DataReader.
That because even if any "default" DataReader ("UDPDataReader", "DefaultDataReader") is not sufficient for a purpose (or even faulty), it will be possible to replace it by a specialized one.
This one can read other sources (dummy byte array or whatever) or add further checks (e.g. for "ready()") on the stream.

It's indeed very convenient to have the current constructors ("SentenceReader(InputStream)" and "SentenceReader(DatagramSocket)"), but for further extensibility the caller should be able to provide it's own reader.

If there had been a constructor with a custom DataReader ("SentenceReader(DataReader)"), I never opened this issue in the first place, since I simply had provided a DataReader, which is checking "ready()" on the stream.

goyalnaresh, 2018-07-12:
Hi @ktuukkan 

I have the same case as @GPSHansl 

Below is the use case:
I am trying to handle the case when StreamReader is not ready then to avoid calling read() method. This is spamming logs continuously with 
Underlying input stream returned zero bytes.

ktuukkan, 2018-07-18:
Ok, thanks for your input @GPSHansl & @goyalnaresh. I'll re-open this and fix the issue with `ready()` and also consider adding the constructor too as it wouldn't hurt anyways.

ktuukkan, 2018-12-29:
`AbstractDataReader` is now public and may be given as a constructor param to `SentenceReader`. The `DataReader` interface has been removed as it seemed unnecessary.

closed 2018-12-29 by ktuukkan