ktuukkan/marine-api #26: Range checking when accessing Sentence fields

GPSHansl, 2015-01-08:
`getStringValue()` should throw a `DataNotAvailableException` whenever the index is not within the `fields` range. Currently it is throwing an `IndexOutOfBoundsException`.

BTW "CodeSmell": `hasValue(int index);` uses Exception-oriented-programming...

ktuukkan, 2015-01-08:
IndexOutOfBoundsException is perfectly valid as it tells that the requested field does not exist (i.e. you made a mistake while writing a parser). DataNotAvailableException tells you that the field exists but is empty (i.e. you didn't make a mistake, advices end-user to handle it somehow). Also, the number of fields in each sentence is usually very well known and the aim is to hide indexes from the end-user anyways. There are also only few sentences with variable length or that have different length between NMEA versions.

About exceptions vs. default values, there's also issue #18. After having thought this over couple of times already, defining a default value for each data type isn't that straight-forward. For example, think of latitude and longitude with default values, say 0.0. Does it mean you're at the equator or that there is no GPS fix? Also with speed, would zero mean "not known" or "no movement"? The same goes pretty much with returning NULL, because it's a bad habit in general and you can't even assign it to primitive types. And it if you did, you would have an exception anyways. Therefore, throwing an exception indicates unambiguously that the value is missing and you should do something else.

I see you're not a fan of exceptions, but as a work-around you may luckily replace any parser in SentenceFactory with a version that supresses the exceptions if you want.

GPSHansl, 2015-01-09:
What about a partial read Sentence, e.g. "$GPGGA,"?
```java
        final GGASentence s = (GGASentence)SentenceFactory.getInstance().createParser("$GPGGA,");
```

Following assertion works flawlessly, wheras I think it's wrong (isn't it?):
```java
        assertTrue(s.isValid());
```

Following assertion will throw an "IndexOutOfBound" instead of returning a correct "INVALID"
```java
        assertEquals(GpsFixQuality.INVALID, s.getFixQuality());
```

ktuukkan, 2015-01-09:
The isValid() method validates the sentence in terms of correct message format, but it doesn't validate each field at once because that would make general validation more complex. This is why each field is validated individually by parser setters and getters. The incomplete sentence String in your example is indeed a valid sentence (no formatting errors), but of course it is incomplete as GGA message. Thus, IndexOutOfBoundsException is again most feasible result because we don't want to go guessing what the value might be when a field is not present.

If you wanted to create empty GGA sentence, you should use
<pre>
..createParser(TalkerId.GP, SentenceId.GGA);
</pre>

closed 2015-01-12 by GPSHansl