ktuukkan/marine-api #32: MWDParser/Sentens and MDAParser

ritchieGitHub, 2015-01-20:


ktuukkan, 2015-01-20:
Thanks, these would be nice addition.

ritchieGitHub, 2015-01-24:
Ok, i am not so gold with gut yet. But i will try to do it tomorrow.

ritchieGitHub, 2015-01-24:
Sorry github  is also New for me...

ritchieGitHub, 2015-01-25:
ok, 3 pull requests ready (hopefully)

ktuukkan, 2015-01-25:
Hi, thanks for the parsers. Just reviewed your pull requests #34 and #35 and otherwise looking good, but I think the getters are bit too strict with units and possibly missing values. For example, what if some device produced sentences with empty unit field or values given in different units? I know, the NMEA spec may say celcius and mercury inches, but in reality you'd never know what might be in there.. Thus, I'd drop the whole if-clause in each getter and let end-users deal with checking the units if needed.

Also, returning NaN is not marine-api convention at the moment as it usually results in more complex error handling. It's quite close to returning nulls, i.e. requires lots of checking with if-clauses instead one clean try-catch.

Corresponding setters would also be good to add in order to allow sentence encoding from a scratch.

ritchieGitHub, 2015-01-25:
I think it is best when the code is as simmular as possible. So do it how you want it. But i think the NaN stuft is i important in this case because of many intendet empty fields (almoust no Sensor sends all the data). This way i just have to call the getter, and if it is not NaN it was available. Setter i did not make because it is explizitly not recomended to generate such sentences. XDR should be used in stead.

ritchieGitHub, 2015-01-25:
Another thing ;-) how mich Chance is there for a new release soon?

ktuukkan, 2015-01-25:
Ok, I'll merge and finish them soon. I'm waiting for some other additions before the next release, but hopefully quite soon(ish). In the meanwhile, you can use the snapshots (see readme.md).

ritchieGitHub, 2015-01-26:
ok, i am using the snapshot now. thanks!

ktuukkan, 2015-02-07:
Hi, any ideas what's the correct interpretation for barometric pressure fields MDA? The current implementation assumes the other set of fields to be in mercury inches (Hg), but test data shows that there may be other units as well. I'm thinking, maybe the field getters/setters should be called "primary" and "secondary" or something similar (e.g. getPrimaryBarometricPressure() etc).

ritchieGitHub, 2015-02-13:
just scanned a littel throu the examples i collected, and it seems like the unit of the first field varries but the unit of the second seems to be always bars. probably best to be on the save side and do as you say name them primary and secondary. I will make a new pull request for that.

ritchieGitHub, 2015-02-13:
ok pull request created, with a unit test for other unit types

closed 2015-02-14 by ritchieGitHub