ktuukkan/marine-api #38: Usages of SentenceReader and UDPDataReader confusing

minghia, 2015-02-03:
Is it intended to use both SentenceReader  and UDPDataReader  in the one application? I'm a little confused as SentenceReader can create  an UDPDataReader  and SentenceReader creates a UDPDataReader if it is instantiated with a DatagramSocket. Would it be possible to have an example of creating a SentenceReader with a DatagramSocket? Thanks in advance.

Tony

ktuukkan, 2015-02-04:
UDPDataReader is used only internally (visible only in package scope), so you should just give DatagramSocket to SentenceReader, register some listeners and start the reader.

minghia, 2015-02-05:
Thanks for that. I sort of work that out by trying to use SentenceReader. By the way does the system handle the concating of messages? For example can you use this system to concat a VLW and VHW message and send them in one message? I tried this in Java by concating the two messages but had no success in receiving the combined message.

ktuukkan, 2015-02-05:
According to NMEA spec each message is terminated by newline and thus SentenceReader and parsers handle only one message at a time. So if you need to concat messages, you should separate them with Sentence.TERMINATOR.

minghia, 2015-02-05:
I create each of the message with the appropriate parser and then I tried to concatenate the two messages with System.arraycopy after converting each Sentence to a series of bytes the lengths of the concated message look okay. Does calling toSentence() remove the terminator? It seems to be missing when I look at the bytes I am writing down the socket.

ktuukkan, 2015-02-05:
Yes, current implementation of toSentence() excludes the terminator chars (it probably should include it, though).

closed 2015-02-21 by ktuukkan