ktuukkan/marine-api #39: getConstructor reflection on enum TalkerId

jasonhector, 2015-02-10:
Hi I am trying to use your nmea library on a kindle and run it as a java applet [or kindlet] . When I debug the code seems to stop in SentenceFactory.registerParser() at the "parser.getConstructor(TalkerId.class)" Im not a reflections expert and even after googling cant seem to determine what "parser.getConstructor(TalkerId.class)" is doing and why it works [since getConstructor for enums is not allowed [according to stackexchange]]. 
I tried catching other exceptions but nothing is caught. 

Could you tell my why this method would hang and what it is doing? Is there another way to implement the functionality that this method is doing? [get around reflections as my guess is that the kindlet doesnt support this]

Thanks
Jason

ktuukkan, 2015-02-10:
Hi, the failing code is checking if parser to be registered has constructor that takes TalkerId enum as parameter. It seems Kindle's Java is based on older Java ME (a subset of full Java SE) which in turn has the older version of class Class (not 100% sure, but as per quick googling it seems so). The signature of getConstructor() is slightly different between these versions.

[Java ME](http://docs.oracle.com/javame/config/cdc/ref-impl/pbp1.1.2/jsr217/java/lang/Class.html#getConstructor%28java.lang.Class%5B%5D%29):
<pre>getConstructor(Class[] parameterTypes)</pre>
vs. [Java SE](http://docs.oracle.com/javase/1.5.0/docs/api/java/lang/Class.html#getConstructor%28java.lang.Class...%29):
<pre>getConstructor(Class&lt;?&gt;... parameterTypes)</pre>

So, it's basically the same but with different syntax. I'll fix it compatible with the old API and let's see if it helps.

jasonhector, 2015-02-11:
Hi, Thanks for the feedback so far.
I changed to your fix but the code still hangs at the same place [see below]. Note that there is no compile errors, and that the kindlet starts fine but then hangs at "parser.getConstructor(TalkerId.class)"

Here is my/your code[I littered with debug print statements]:
	public void registerParser(String type, Class<? extends SentenceParser> parser) {
		log("SentenceFactory::registerParser() for ".concat(type));
		try {
			parser.getConstructor(new Class[]{String.class});
			log("SentenceFactory::registerParser: post getConstructor for String");
			//FIXME this seems to hang here!!!
			parser.getConstructor(new Class[]{TalkerId.class});
			log("SentenceFactory::registerParser: post getConstructor for TalkerId");
			parsers.put(type, parser);
			log("SentenceFactory::registerParser: post parser put()");
		} catch (SecurityException e) {
			String msg = "Unable to register parser due security violation";
			log("SentenceFactory::registerParser: SecurityException:".concat(msg));
			throw new IllegalArgumentException(msg, e);
		} catch (NoSuchMethodException e) {
			String msg = "Required constructors not found; SentenceParser(String), SentenceParser(TalkerId)";
			log("SentenceFactory::registerParser: NoSuchMethodException:".concat(msg));
			throw new IllegalArgumentException(msg, e);
		}catch (IllegalArgumentException e){
			log("SentenceFactory::registerParser: IllegalArgumentException:");
		}
	}

and here is the debug logs:
[root@kindle tmp]# tail -f dbg.log
11-2-2015 09:17:35.373::AbstractDataReader::run: post ActivityMonitor
11-2-2015 09:17:35.380::SentenceReader::start: reader thread started!
11-2-2015 09:17:35.381::sr start done
11-2-2015 09:17:35.526::SentenceFactory::getInstance()..
11-2-2015 09:17:35.526::SentenceFactory::SentenceFactory()..
11-2-2015 09:17:35.702::SentenceFactory::registerParser() for APB
11-2-2015 09:17:35.743::SentenceFactory::registerParser: post getConstructor for String

jasonhector, 2015-02-11:
After digging a bit myself, the supported JRE is 1.4 [or earlier] on the kindle for the cvm. So no typed parameters. I see the 0.1 release requires java 1.5. <Is this a dead-end>

Would it be substantially challenging to have the library compatible with JRE 1.4? What would need to be done? I dont mind forking and doing the work, but if its a complete redesign of your library then it might not be worth the trouble.
Thanks

ktuukkan, 2015-02-11:
This is what I was suspecting after having second thought. Making the library compatible with 1.4 is probrably quite an effort as generics have been used in many places. Maybe not a complete redesign, but in some parts might get a bit tricky (e.g. AbstractSentenceListener).

jasonhector, 2015-02-11:
Hi. One of.my earlier kindle project used a library called retroweaver which allows the use of later java jdks and it gets past the failing code. So this is promising. Gonna run some further tests tonight.

closed 2015-02-12 by jasonhector