ktuukkan/marine-api #42: SentenceReaderTest.testExceptionListener()

gump67, 2015-02-25:
Need to close your reader once you are finished, otherwise the thread will keep running in the background.
....
		reader.start();

		try {
			stream.close();
		} catch (IOException e) {
			fail("failed to close stream");
		}
		>>>reader.stop();
	}

ktuukkan, 2015-02-26:
Thanks, nice catch. I'll check this out soon.

gump67, 2015-04-09:
I apologize, after reviewing this issue further I realized that there is actually a major problem in the source code as well.

A few lines up, you will notice that the test code expects the stream.close() to throw an exception.  This exception is outlined by your anonymous class  with the overridden method onException. However, this exception is not invoked and as such is missed entirely.

If you are applying TDD, first you need to create a custom exception (create a separate class) and have a 'was called' get/setter.   Add the set call accordingly to the onException operation.  Then you will need to check to see if this exception was called at the end of the code, otherwise throw a failure.

If this is confusing, let me know and I can clarify at a later date.  :)

gump67, 2015-04-09:
Keep in mind you will still need to fix the source code after adding the Test code.

ktuukkan, 2015-04-09:
Actually, it seems this test was outdated now that DefaultDataReader performs ready() check for InputStream before trying to read from it (this was added recently). That is, with this check in place closing the stream does not anymore result in exception thrown as it used to and thus onException() never got called.

Refactored the test for now so that handleException() is called directly. Further tests with DataReader actually throwing an exception remains still to be figured out (if really needed).

closed 2015-07-04 by ktuukkan