ktuukkan/marine-api #43: ClassCastException

Seraphis, 2015-03-18:
I get 
ClassCastException: java.lang.Class cannot be cast to java.lang.reflect.ParameterizedType
at AbstractSentenceListener.java:52

I have pretty similar code to your example at the end of https://github.com/ktuukkan/marine-api/blob/master/src/main/java/net/sf/marineapi/example/SentenceListenerExamples.java file.

ktuukkan, 2015-03-18:
Hi,
what's the generic type you are using when extending AbstractSentenceListener? It should be one of the actual sentence interfaces (e.g. GGASentence, but any interface GGASentence extends won't work at the moment).

Seraphis, 2015-03-19:
I'm using GGASentence:
public class GGAListener extends AbstractSentenceListener<GGASentence>

ktuukkan, 2015-03-20:
I'm unable to reproduce this, could you paste a snippet of your code?

Seraphis, 2015-03-20:
```java
public class GGAListener extends AbstractSentenceListener<GGASentence> {
		private BluetoothNMAEListener bluetoothNMAEListener;

		public GGAListener(BluetoothNMAEListener bluetoothNMAEListener) {
			super();
			this.bluetoothNMAEListener = bluetoothNMAEListener;
		}
}
```

ktuukkan, 2015-04-01:
Your code is ok, but I still haven't been able to reproduce this on JDK, maybe Android specific? Which Java version are you using?

Seraphis, 2015-04-03:
There might be a problem with proguard. Debug apk was working fine. I am still waiting for test results.

ktuukkan, 2015-07-15:
Closing the issues as related in external tools.

closed 2015-07-15 by ktuukkan