ktuukkan/marine-api #44: Incorrect transform to java.util.Date

flawlesscoder, 2015-04-01:
Month must be one less for the java.util.Calendar.
Change it on this (net.sf.marineapi.nmea.util.Date):
```
    public java.util.Date toDate() {
        GregorianCalendar cal = new GregorianCalendar();
        cal.set(Calendar.YEAR, getYear());
        cal.set(Calendar.MONTH, getMonth() - 1);
        cal.set(Calendar.DAY_OF_MONTH, getDay());
        cal.set(Calendar.HOUR_OF_DAY, 0);
        cal.set(Calendar.MINUTE, 0);
        cal.set(Calendar.SECOND, 0);
        cal.set(Calendar.MILLISECOND, 0);
        return cal.getTime();
    }
```

Forsety, 2015-04-01:
This issue is already fixed in my pull request ktuukkan/marine-api#24.

ktuukkan, 2015-04-01:
Yes this should be fixed already, please check if you have fetched latests changes or snaphot.

closed 2015-04-10 by ktuukkan