ktuukkan/marine-api #48: Thread.sleep(50) in AbstractDataReader.java makes files parsing slow

adrianulbona, 2015-07-13:
I tested the parser on some large files and it looks like a SentenceReader is quite slow. I suppose that the sleep is required when the UDPDataReader implementation is used, but for the DefaultDataReader it makes sense to call Thread.sleep()?

ktuukkan, 2015-07-15:
Hi, it's there mainly to prevent reader loop from eating all CPU. I think it would be feasible to make it configurable, if not getting rid of it totally if possible.

adrianulbona, 2015-07-15:
Thanks for answer. I will think of a solution to make the sleep time configurable and I will send you a pull request. 

By the way, this library is really helpful. :)

ktuukkan, 2015-07-15:
Thanks, looking forward to it. I'm glad you like it :)

adrianulbona, 2015-07-20:
Added pull request #49. As everything is running on a thread, the CPU overhead will be reasonable, the OS will handle the CPU load.

ktuukkan, 2015-10-30:
Added the feature in master now, resolving the issue as I proposed earlier in pull-req #49 (closed).

closed 2015-10-30 by ktuukkan