ktuukkan/marine-api #53: Misunderstanding of String.format for float

azaraphel, 2016-04-12:
In #toString of Position, it is used format with "%02.04f" and "%03.04f" pattern. It seems you want a left padding like with the "%02d" pattern. But with 'f', the left numerical value indicates the total number of characters.
Try this :
System.out.println(String.format("%02.04f", 1.23456F));
System.out.println(String.format("%04.04f", 1.23456F));
System.out.println(String.format("%05.04f", 1.23456F));
System.out.println(String.format("%06.04f", 1.23456F));
System.out.println(String.format("%07.04f", 1.23456F));
System.out.println(String.format("%08.04f", 1.23456F));
System.out.println(String.format("%019.04f", 123.45F));

ktuukkan, 2016-04-13:
Good catch, I'll check it out soon. Thanks!

ktuukkan, 2016-04-17:
Fixed using DecimalFormat; left padding to 2 or 3 digits with 7 decimals.

closed 2016-04-30 by ktuukkan