ktuukkan/marine-api #55: java.lang.IllegalArgumentException: Parser for type 'TXT' not found

wureka, 2016-04-29:
Hi, my code happened the below exception. Would you please tell me how to fix ?

My code:
`
public class GpsTest implements SatelliteInfoListener {

    SentenceReader reader;
    SatelliteInfoProvider provider;
    public void GpsTest0(File file) throws IOException {
        // create sentence reader and provide input stream
        InputStream stream = new FileInputStream(file);
        reader = new SentenceReader(stream);
        // create provider and register listener
        provider = new SatelliteInfoProvider(reader);
        provider.addListener(this);
        reader.start();
    }
    public static void main(String[] args) throws IOException {
        File f= new File("/dev/ttyS0");
        GpsTest tt = new GpsTest();
        tt.GpsTest0(f);
    }
`

Exception is below:

`Apr 29, 2016 1:38:13 PM net.sf.marineapi.nmea.io.AbstractDataReader run
WARNING: Data read failed
java.lang.IllegalArgumentException: Parser for type 'GBS' not found
        at net.sf.marineapi.nmea.parser.SentenceFactory.createParserImpl(SentenceFactory.java:212)
        at net.sf.marineapi.nmea.parser.SentenceFactory.createParser(SentenceFactory.java:95)
        at net.sf.marineapi.nmea.io.AbstractDataReader.run(AbstractDataReader.java:98)
        at java.lang.Thread.run(Thread.java:745)

Apr 29, 2016 1:38:13 PM net.sf.marineapi.nmea.io.AbstractDataReader run
WARNING: Data read failed
java.lang.IllegalArgumentException: Parser for type 'TXT' not found
        at net.sf.marineapi.nmea.parser.SentenceFactory.createParserImpl(SentenceFactory.java:212)
        at net.sf.marineapi.nmea.parser.SentenceFactory.createParser(SentenceFactory.java:95)
        at net.sf.marineapi.nmea.io.AbstractDataReader.run(AbstractDataReader.java:98)
        at java.lang.Thread.run(Thread.java:745)`

GPS output as below:

`

$GPVTG,,,,,,,,,N*30
$GPGGA,,,,,,0,00,99.99,,,,,,*48
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30
$GPGSV,1,1,00*79
$GPGLL,,,,,,V,N*64
$GPGST,,0.0000,,,,3750561,3750561,3750561*7A
$GPZDA,,,,,00,00*48
$GPGBS,,,,,,,,*41
$GPTXT,01,01,01,NMEA unknown msg*58
$GPTXT,01,01,01,NMEA unknown msg*58
$GPTXT,01,01,01,NMEA unknown msg*58
$GPTXT,01,01,01,NMEA unknown msg*58
$GPTXT,01,01,01,NMEA unknown msg*58

`

ktuukkan, 2016-04-30:
Hi! This is simply due that CBS and TXT sentences are not currently supported, so SentenceFactory throws an error because it cannot handle them. If you don't care about these sentences, you can simply ignore this, it's basically just a warning.

If you need the information transmitted by these sentences, you may implement the parsers and register them in SentenceFactory before starting the reader, see SentenceFactory javadocs for step-by-step instructions. Also, if you'd also wish to contribute these parsers in the library, please open a pull-request for each.

nm3210, 2016-11-28:
Hey @ktuukkan, I think I'm having an issue related to setting an exception listener in the v0.9.0 release (not sure if anything has been fixed in 0.10.0 for it). Per your recommendation, I am trying to suppress errors from the AbstractDataReader for bad/unrecognized lines in the nmea file.

The code seems straightforward enough:
```nmeaReader = new SentenceReader(stream);
nmeaReader.addSentenceListener(this);
nmeaReader.setInterval(50);
nmeaReader.setExceptionListener(new ExceptionListener(){
	@Override
	public void onException(Exception e) {
		System.err.println("onException...please print me");
	}});
nmeaReader.start();
```

However, I get the following out with no acknowledgement that my new ExceptionListener was used:

> Nov 28, 2016 9:25:15 AM net.sf.marineapi.nmea.io.AbstractDataReader run
> WARNING: Data read failed
> java.lang.IllegalArgumentException: Parser for type 'DAS' not found
> 	at net.sf.marineapi.nmea.parser.SentenceFactory.createParserImpl(SentenceFactory.java:212)
> 	at net.sf.marineapi.nmea.parser.SentenceFactory.createParser(SentenceFactory.java:95)
> 	at net.sf.marineapi.nmea.io.AbstractDataReader.run(AbstractDataReader.java:98)
> 	at java.lang.Thread.run(Unknown Source)

Am I just doing something wrong? Thanks!!

ktuukkan, 2016-12-03:
Hi @nm3210, in v0.9 ExceptionHandler is called only when actual read errors are thrown while reading the stream or UDP socket. In 0.10 however, exceptions thrown by SentenceFactory are also passed to ExceptionHandler to enable suppressing them.

My bad, should have mentioned earlier that this works only in 0.10 snapshot.

ktuukkan, 2017-01-28:
Created separate tasks for adding support for GBS, GST and TXT sentences, thus closing this one.

closed 2017-01-28 by ktuukkan

nadjibmaka, 2017-06-07:
still no parser for TXT ? :)