ktuukkan/marine-api #56: PositionProvider exception if RMC is missing FAA mode field

Turix, 2016-06-22:
Hi,

I've been having a bit of a play around with MarineAPI and I think I've found an issue:

If I try to use a PositionProvider with RMC sentences that do not contain the optional FAA Mode field, the following exception occurs:

```
Jun 22, 2016 1:20:07 PM net.sf.marineapi.nmea.io.SentenceReader fireSentenceEvent
WARNING: Exception caught from SentenceListener
java.lang.IndexOutOfBoundsException: Index: 11, Size: 11
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.get(ArrayList.java:429)
	at net.sf.marineapi.nmea.parser.SentenceParser.getStringValue(SentenceParser.java:430)
	at net.sf.marineapi.nmea.parser.SentenceParser.getCharValue(SentenceParser.java:375)
	at net.sf.marineapi.nmea.parser.RMCParser.getMode(RMCParser.java:109)
	at net.sf.marineapi.provider.PositionProvider.isValid(PositionProvider.java:147)
	at net.sf.marineapi.provider.AbstractProvider.validate(AbstractProvider.java:239)
	at net.sf.marineapi.provider.AbstractProvider.sentenceRead(AbstractProvider.java:216)
	at net.sf.marineapi.nmea.io.SentenceReader.fireSentenceEvent(SentenceReader.java:206)
	at net.sf.marineapi.nmea.io.AbstractDataReader.run(AbstractDataReader.java:90)
	at java.lang.Thread.run(Thread.java:745)
```

I have tried to confirm this in isolation of my application by running the a sample file from http://www.gpsinformation.org/dale/nmea.htm#stream (Garmin g12) through the PositonProviderExample. Whereas other samples on the linked page that contain the field work as expected.

```
$GPRMC,183729,A,3907.356,N,12102.482,W,000.0,360.0,080301,015.5,E*6F
$GPRMB,A,,,,,,,,,,,,V*71
$GPGGA,183730,3907.356,N,12102.482,W,1,05,1.6,646.4,M,-24.1,M,,*75
$GPGSA,A,3,02,,,07,,09,24,26,,,,,1.6,1.6,1.0*3D
$GPGSV,2,1,08,02,43,088,38,04,42,145,00,05,11,291,00,07,60,043,35*71
$GPGSV,2,2,08,08,02,145,00,09,46,303,47,24,16,178,32,26,18,231,43*77
$PGRME,22.0,M,52.9,M,51.0,M*14
$GPGLL,3907.360,N,12102.481,W,183730,A*33
$PGRMZ,2062,f,3*2D
$PGRMM,WGS 84*06
$GPBOD,,T,,M,,*47
$GPRTE,1,1,c,0*07
$GPRMC,183731,A,3907.482,N,12102.436,W,000.0,360.0,080301,015.5,E*67
$GPRMB,A,,,,,,,,,,,,V*71
```

From what I can work out, the culprit is `RMCParser#getMode` being called without either the field existing (and hence being loaded by the `SentenceParser` constructor) or `RMCParser#setMode` being called (which explicitly sets the length of `fields` before adding a mode) first. Which is causing the above `IndexOutOfBoundsException` to occur when `SentenceParser#getStringValue` attempts to access the position of the field.

ktuukkan, 2016-06-26:
Hi, it's quite likely this hasn't just been taken into account while writing the provider as NMEA has quite many fields that may or may not be there. Thanks for reporting!

ktuukkan, 2016-07-04:
Fix deployed in snapshots repo, please verify and close the issue if all is good.

Turix, 2016-07-05:
That appears to have solved the issue, thanks!

closed 2016-07-05 by Turix