ktuukkan/marine-api #58: Parsers for GNS Sentences

srehlig, 2016-09-29:
Hi,

the library looks good to me. We are integrating a u-blox GNSS receiver into an android platform and are evaluating marine-api.

What about extending the library with parsers for http://www.trimble.com/OEM_ReceiverHelp/V4.44/en/NMEA-0183messages_GNS.html ?

Are there any plans or is that already possible and i missed it?

Thanks,
Stefan

ktuukkan, 2016-09-30:
Hi,
currently there's no implementation for GNS but it seems quite straightforward to add. I can look into it but unfortunately cannot promise very tight schedule. Another option would be that you implement it yourself and register the parser in SentenceFactory at runtime (see SentenceFactory javadocs). In this case you may also contribute the code in main repo if you wish.

ktuukkan, 2016-11-06:
Support for GNS added and now available in latest snapshot.

closed 2016-11-06 by ktuukkan