ktuukkan/marine-api #61: Support for seatalk Stalk sentence, and signalk

rob42, 2017-01-24:
Over in the signalk.org project we have been using marineapi in the java server implementation, and its great!

We would like support for the seatalk over NMEA Stalk sentences and wonder if thats in scope for you?

Also @ktuukkan, we talked some years ago now about signalk support in marineapi, that would be great too, and make marineapi the premier way to do NMEA to IoT and internet. Your thoughts?

ktuukkan, 2017-01-25:
Hi @rob42, nice to hear from you, I'm glad it works well for you!

I'm not that familiar with SeaTalk but basic support for $STALK would seem pretty easy to implement as it's mainly a variable number of hex fields. Would this be sufficient for you as a start? Further decoding of messages would seem to require more knowledge on the format and could probably work in the same way as with AIS messages, i.e. having an extra layer of parsers handling each SeaTalk message delivered by common $STALK parser. However, the current AIS implementation is not quite finished yet, so I'd like to finish that first before going in bigger new feature. And yeah, I know, I've been very lazy on it for the last year or so.. It's mainly lack of time than interest.

SignalK support is also interesting idea and I should probably see what's happened in your project after we last discussed it. You may already have some ideas for it, but I guess it would basically mean marine-api to parse and output SignalK JSON?

rob42, 2017-01-29:
Re $STALK, yes that sounds good.
Signalk support would ideally provide translation from NMEA direct to signalk, and potentially in reverse, although this is more complex as you generally need a variety of signalk data values for a single NMEA message.
You can get a reasonable overview at signalk.org, and via github, The best place to discuss is on slack, there is a link to let you join on the signalk.org page. There is a channel for $STALK at https://signalk-dev.slack.com/messages/stalk/

Ive got quite a bit of code to convert from your library to signalk in the various signalk-java-* repos on github which you are very welcome to use (although its in need of a decent clean up)

rob42, 2017-01-29:
https://github.com/SignalK/signalk-core-java/blob/master/src/main/java/nz/co/fortytwo/signalk/handler/NMEAHandler.java converts from NMEA to a map of signalk values. 
https://github.com/SignalK/signalk-core-java/blob/master/src/test/java/nz/co/fortytwo/signalk/handler/NMEAHandlerTest.java  the associated test.
I have existsing code to convert the flat map to signalk json output.

ktuukkan, 2017-01-30:
Ok, cool. I'll split this in separate issues for SeaTalk and SignalK. Let's record things of importance under those to keep it clear.

ktuukkan, 2017-01-30:
Issues #67 and #68 created.

closed 2017-11-19 by ktuukkan