ktuukkan/marine-api #62: MWDSentence

titio72, 2017-01-27:
Hi,

First of all: great work!

I've fixed a couple of bugs in the MWDParser - the number of fields was wrong (20 instead of 8) and the speed unit also (K instead of N).
If you want you can check out the fix in my clone (tito72/marine-api)

Very minor: in my clone I've added also a new talker (OC) for OpenPlotter.

Cheers

ktuukkan, 2017-01-28:
Hi, thanks @titio72 !

Must have been simple copy/paste mistakes, and looks like the tests are not covering those. OC seems non-standard, but I believe it's also safe to include in the lib to provide support for OpenPlotter.

Could you open a pull-request so it's easier to merge your changes?

ktuukkan, 2017-01-29:
PR merged, thanks!

closed 2017-01-29 by ktuukkan