ktuukkan/marine-api #68: Support for SignalK JSON

ktuukkan, 2017-01-30:


ktuukkan, 2017-01-30:
NMEA to a map of SignalK values:
https://github.com/SignalK/signalk-core-java/blob/master/src/main/java/nz/co/fortytwo/signalk/handler/NMEAHandler.java

Associated tests:
https://github.com/SignalK/signalk-core-java/blob/master/src/test/java/nz/co/fortytwo/signalk/handler/NMEAHandlerTest.java

ahsanilya-s, 2025-08-21:
Thanks for opening this feature request. Adding SignalK JSON support (both NMEA → SignalK and ideally SignalK → NMEA) would really improve interoperability. The linked NMEAHandler and its tests are a good starting point. I’d be interested to know if there are any current limitations or known challenges when mapping complex NMEA messages into multiple SignalK values. Happy to help test or review if progress is made.