ktuukkan/marine-api #78: Usage of instanceof to read sentences

russellhoff, 2017-12-21:
In your examples, you read NMEA sentences comparing their String IDs that way:

```
	@Override
	public void sentenceRead(SentenceEvent pEvent) {
		Sentence s = pEvent.getSentence();
		String sentenceId = s.getSentenceId();
		if(sentenceId.equalsIgnoreCase("GGA")) {
			GGASentence gga = (GGASentence) s;
			// ...
		}else if(sentenceId.equalsIgnoreCase("GLL")) {
			GLLSentence gll = (GLLSentence) s;
			// ...
		}else {
			// ...
		}

	}
```

My point is that it's better to use `instanceof``instead of equalising strings:
```
	@Override
	public void sentenceRead(SentenceEvent pEvent) {
		Sentence s = pEvent.getSentence();
		if(s instanceof GGASentence) {
			GGASentence gga = (GGASentence) s;
			// ...
		}else if(s instanceof GLLSentence) {
			GLLSentence gll = (GLLSentence) s;
			// ...
		}else {
			// ...
		}

	}
```

This should work, as every "leaf" sentence, e.g. GGASentence or GLLSentence, inherits from a root interface called `net.sf.marineapi.nmea.sentence.Sentence`.

Don't you think is it a better choice?

ktuukkan, 2017-12-21:
@russellhoff The most preferred way is to extend `AbstractSentenceListener` to create separate handlers for each sentence type.

Using `instanceof` is often discouraged, but ofcourse I may be used like you suggested.

russellhoff, 2017-12-21:
Thanks, [here](http://www.javapractices.com/topic/TopicAction.do?Id=31) is an example of the disadvantages of the usage of `instanceof`

closed 2017-12-21 by russellhoff