ktuukkan/marine-api #79: SentenceReader.start() after setInputStream(...) throws exception; no events

akidee, 2018-03-03:
This is how I use SentenceReader in Android for GGA sentences:

`final static SentenceReader SR = new SentenceReader(new ByteArrayInputStream(new byte[0]));
{
    SR.addSentenceListener(new AbstractSentenceListener<GGASentence>() {

        @Override
        public void sentenceRead(GGASentence sentence) {

            Log.d(TAG, "sr.sentenceRead("+sentence+")");
        }
    }, SentenceId.GGA);
    SR.start();
}
final static Charset ASCII = Charset.forName("US-ASCII");

public void feedNmeaSentence2(long timestamp, String message) {

    SR.setInputStream(new ByteArrayInputStream(message.getBytes(ASCII)));
    try {
        SR.start();
    }
    catch (IllegalStateException e) {

    }
}`

Calling SR.start() the first time is no problem. But calling it after setInputStream() (with the doc saying 

> Sets the InputStream to be used as data source. If reader is running, it is first stopped and you must call start() to resume reading.

throws an IllegalStateException. Only the first call to start() seems to work and I can receive one GGASentence. Calling start() only one time and omitting the call after setInputStream() results in no events.

ktuukkan, 2018-03-09:
Thanks for reporting, this is quite likely a timing problem, will have a look on it. On the other hand, it looks like you don't actually need `SentenceReader` for the second message as it is already a String. Just parse it with `SentenceFactory` and check the sentence id of returned `Sentence`, and then cast it into corresponding sentence interface.

ktuukkan, 2020-07-02:
Closing this because I'm affraid I don't fully understand your use case and thus also unable to reproduce the problem. Feel free to open new issue if needed.

closed 2020-07-02 by ktuukkan