ktuukkan/marine-api #91: How to Parser type Message $PVTM0 ?

DatangYa, 2018-10-03:
sample  $PVTM0 ,ID,ddmmyyyy,hhmmss,tt.ttt,h.hhh,D,vv.v,SSSS*hhCL

ktuukkan, 2018-10-05:
From [TimeMaster Operating Manual](https://www.scribd.com/doc/243766235/TideMaster-Operating-Manual):

Tide data, same for Pressure or Radar measurment

`$PVTM0,ID,ddmmyyyy,hhmmss,tt.ttt,h.hhh,D,vv.v,SSSS*hhCL`

- ID Station ID
- Date ddmmyyyy
- Time hhmmss
- Tide Height tt.ttt
- Tide height stdev h.hhh
- Tide height units D (0,1,2) (dBar, Metres, Feet)
- Battery voltage vv.v
- Status
- *hh checksum of the above
- CL <cr><lf> end of string

DatangYa, 2018-10-08:
registerParser("VTM", VTMParser.class);
ex : $PVTM0 ? How to registerParser ?

ktuukkan, 2018-10-09:
`registerParser("VTM0", VTM0Parser.class);`

closed 2018-11-10 by ktuukkan