ktuukkan/marine-api #94: how to get position to a string sentence ?

hananabilabd, 2018-11-22:
um wondering how to get get position to a string sentence ?

ktuukkan, 2018-11-23:
Hi,

the [PositionSentence](http://static.javadoc.io/net.sf.marineapi/marineapi/0.10.0/net/sf/marineapi/nmea/sentence/PositionSentence.html) interface allows you to get and set the [Position](http://static.javadoc.io/net.sf.marineapi/marineapi/0.10.0/net/sf/marineapi/nmea/util/Position.html) object in sentences. This interface is implemented by sentences containting the current position, for example GLL, RMB etc.

hananabilabd, 2018-11-23:
I use shareGPS application on android to get GPS data from mobile to my laptop through serial ports so when I used sqentence listener .. I tried everything but something wrong with reading the inputStream 
But I then handled the Bytes and I got the string sentences from the serial ... 
So my question is how to to convert the string sentence to instance to get position and speed

hadialbadaoui, 2018-11-23:
Hi, can u send for me the string sentence as an example?
I need to know the start bit, end bit, etc... to clarify the sentence then I will search for a solution
Thank you, best regards.

ktuukkan, 2018-11-23:
> how to to convert the string sentence to instance to get position and speed

`SentenceFactory.getInstance().createSentence(nmea);` and then check the type of returned `Sentence` with `getSentenceId()` and cast it to corresponding interface, for example "GGA" to `GGASentence`.

hadialbadaoui, 2018-11-28:
Hello Hanna, 
please I need to know if my answer helped you.
if not please tell me to discuss the problem.
Thank you.

hananabilabd, 2018-11-28:
@hadialbadaoui  Thanks a lot for explanation
I guess I was nor going that deep into the serial frame 
i was in the layer above using rxtx library in java 
and eventually i got a solution to my problem 
but can i ask you one more question >>
How can i get the velocity from the Nmea sentence ?

hadialbadaoui, 2018-11-29:
How can I get the velocity from the NMEA sentence?
WCV - Waypoint Closure Velocity
              1    2  3     4
               |     |   |      |
 $--WCV,x.x,N,c--c*hh<CR><LF>
Field Number:
1- Velocity
2- N = knots
3- Waypoint ID
4- Checksum
Check this URL: http://www.catb.org/gpsd/NMEA.html#_wcv_waypoint_closure_velocity
will explain for you all gps receivers messages

thank you,
Best regards.

hananabilabd, 2018-11-29:
Thanks @hadialbadaoui
But i mean how can i get it using this library , is there is a way ?

closed 2018-12-10 by ktuukkan