ktuukkan/marine-api #99: Parsing of Latitude and Longitude

hlaurent, 2019-02-11:
The parser fails to parse latitude and longitude that do not have leading zeros like:
$GPGLL,4426.291992,N,159.931976,W,143744,A*3B
The parser interpret this as 159°W when it should be 1°59'W.
The parser is waiting for 00159.931976 while it receives 159.931976

This is an error in the parser as the NMEA conventions state :
Where a numeric latitude or longitude is given, **the two digits immediately to the left of the decimal point are whole minutes, to the right are decimals of minutes, and the remaining digits to the left of the whole minutes are whole degrees.**

Eg. 4533.35 is 45 degrees and 33.35 minutes. ".35" of a minute is exactly 21 seconds.
Eg. 16708.033 is 167 degrees and 8.033 minutes. ".033" of a minute is about 2 seconds.

The  parsing functions should be edited in something like:

	protected double parseLatitude(int index) {
		String field = getStringValue(index);
		int lastIndexDot = field.lastIndexOf('.');
		double min = Double.parseDouble(field.substring(lastIndexDot - 2));
		int deg = Integer.parseInt(field.substring(0, lastIndexDot - 2));
		return deg + (min / 60);
	}

protected double parseLongitude(int index) {
		String field = getStringValue(index);
		int lastIndexDot = field.lastIndexOf('.');
		double min = Double.parseDouble(field.substring(lastIndexDot - 2));
		int deg = Integer.parseInt(field.substring(0, lastIndexDot - 2));
		return deg + (min / 60);
	}

ktuukkan, 2019-02-11:
Indeed, you're correct. Thanks for reporting!

ktuukkan, 2019-02-14:
Fix now available in the latest snapshot.

ktuukkan, 2019-02-14:
Re-opened due to issues with 0° lat/lon formatting, for example

- .844957 for 0°00.844957'
- 8.844957 for 0°08.844957'
- 28.844957 for 0°28.844957'

closed 2019-03-16 by ktuukkan