ktuukkan/marine-api #101: removeSentenceListener, by type improvement?

handyli, 2019-03-06:
Hi,
There is addSentenceListener(SentenceListener sl, String type).
Would it make sense to have possibility to remove listener by type? Now only option is:
removeSentenceListener(SentenceListener listener)

I have one listener (tcp port) which I want to receive only some sentences and to be able to change sentences later. Current removeSentenceListener will remove all SentenceListeners.

I have GPS feed coming from serial port. After connection to tcp port for example RMC and GSA would initially sent to tcp client. Client would be able to add or remove sentences it receives. Add works fine, but remove will remove all sentences. Workaround would be to remove all listeners and then to add again those which are still wanted. 

I'm currently using snaphot-0.11.

ktuukkan, 2019-03-08:
Yes, the current implementation removes the listener regardless of types it was added for, so I understand it may be a problem if you've registered the same listener for many types. The suggestion seems feasible and easy to implement, so I don't see problems in adding it.

ktuukkan, 2019-03-16:
@handyli, this is now done in the latest snapshot.

closed 2019-03-23 by ktuukkan