ktuukkan/marine-api #103: openjdk 1.8 net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)         at net.sf.marineapi.nmea.event.AbstractSentenceListener.<init>(AbstractSentenceListener.java:77)

songhaonangit, 2019-04-04:
2019-04-04 18:11:30.388 8411-8411/? E/AndroidRuntime:     at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
2019-04-04 18:11:30.388 8411-8411/? E/AndroidRuntime:     at net.sf.marineapi.util.GenericTypeResolver.resolve(GenericTypeResolver.java:57)
        at net.sf.marineapi.nmea.event.AbstractSentenceListener.<init>(AbstractSentenceListener.java:77)

ktuukkan, 2019-04-08:
Any context or additional info for this?

ktuukkan, 2019-10-19:
Unable to reproduce, thus closing for now.

closed 2019-10-19 by ktuukkan