ktuukkan/marine-api #106: PA flag is always false in AIS messages 1,2,3,18,19 etc

Dazza666, 2019-10-18:
It looks like the PA flag in messages 1,2,3,18,19 etc is always false, i.e. low

For example, if I send a message 1, which contains a PA flag set to high (1), like so:

**!AIVDM,1,1,,A,197vW0G2iTP4fg0Mhr40<P:t0@Ru,0*33**

I get a 'false' when decoding the message.

If, I pump that message into the online decoder at:

https://www.maritec.co.za/tools/aisvdmvdodecoding/

You see that the PA flag is actually 1.

for completeness, here is a the same message with the PA flag encoded to 0

**!AIVDM,1,1,,B,197vW0G2iT04fg0Mhr40<P:t0@:F,0*0B**

Any chance of a quick fix with this one?

Dazza666, 2019-10-18:
I should probably also mention that I believe the fix is actually found in the AIS Message 9:

`fPositionAccuracy = content.getBoolean(**TO**[POSITIONACCURACY]);`

vs the incorrect (in AISPositionReportParser)

fPositionAccuracy = content.getBoolean(**FROM**[POSITIONACCURACY]);

ktuukkan, 2019-10-19:
@Dazza666, thanks for reporting! Looks like an indexing problem and there's also at least one error in the tests that allowed it to pass as correct :/

The class comment in `AISPositionReportParser` says "from 61 to 61" for position accuracy, but in the actual index arrays it's 60 and 61, hence using `TO` instead of `FROM` fixes it. There's also other indices off by one compared to the class comment, so I think it's better check this thoroughly now.

I made a quick-fix as pre your suggestion, try the [snapshot](https://oss.sonatype.org/content/repositories/snapshots/net/sf/marineapi/marineapi/0.12.0-SNAPSHOT/).

Dazza666, 2019-10-21:
I will check this tomorrow and come back to you, I also noticed the indexing positions, but thought perhaps this was intentional for some reason that wasn't clear upon a quick look at the code. Thank you for such a prompt fix!

Dazza666, 2019-10-22:
Right, am happy to report that this issue has been resolved, Position Accuracy now reports correctly.

Note, for anyone using Gradle, you will need to add the following line in order to obtain snaphots

`maven { url 'https://oss.sonatype.org/content/repositories/snapshots' }`

ktuukkan, 2019-10-22:
Ok cool, thanks for testing. I'll still have some further looks on this as the indexing is a bit puzzling. Perhaps @techhead2 still remembers something about this and could enlighten us?

techhead2, 2019-10-22:
Sorry guys, I haven't had a look at the code for years.

ktuukkan, 2019-10-22:
Ok, thanks anyways :)