ktuukkan/marine-api #107: DTE (Data Terminal Equipment) is not decoded in a message 5

Dazza666, 2019-10-22:
DTE (Data Terminal Equipment) is not decoded in a message 5 (AISMessage05Parser) DTE also exists in a message 9; the code for a type 9 **does** expose this field however. 

Message type 18 & 19 (Class B devices) also omit the DTE flag, but this I think is a separate issue, as there are a couple of other fields which don't appear to be decoded here either.

The DTE flag is not decoded in a message 5, There is no variable to hold it, nor does it exist in the index! It must have been accidentally omitted!

DTE is at position 423, 423 according to the comment

So, I think we need to add 

Location in the array:

`private static final int 	DTE			= 16;`

Add location to Array:

```
private final static int[] TO = {
 40, 70, 112, 232, 240, 249, 258, 264, 270, 274, 278, 283, 288, 294, 302, 422, 423};
```
 
Variable:

`private Boolean		fDte;`

Obtaining the flag:

`fDte = content.getBoolean(TO[DTE]);`

Getter:

`public String getDte() { return fDte; }`

lastly in toString():

`result += SEPARATOR + "Dte:      " + fDte;`

ktuukkan, 2019-10-22:
Another good find, thanks! Will have a look on this too.

ktuukkan, 2020-03-15:
Hello @Dazza666, can you confirm the above commit resolves the problem for you?

Dazza666, 2020-03-16:
I should be able to test this for you over the next few days

Dazza666, 2020-06-19:
Hi, sorry this is so late I have only just had a chance to check this out, can confirm that it is fixed on 0.12 snapshot.