ktuukkan/marine-api #113: GSA message filtered timeout

aitov, 2020-04-09:
I'm using BU-353S4 GPS receiver on Mac
There is issue with GSA message which removed by timeout in AbstractProvider:236
`long age = now - se.getTimeStamp();`
			`if (age > 1000) {`
				`return false;`
			`}`
As result I never received SatelliteInfoEvent because GSA message is required part of it (with GSV)
I had to disable this check in order to fix it.
Probably we need increase this timeout for some messages
  
Time in output: 
age: 4463 $GPGSA,A,3,10,27,20,08,21,,,,,,,,3.5,2.6,2.2*3D
age: 3462 $GPGSA,A,3,10,27,20,08,21,,,,,,,,3.5,2.6,2.2*3D
age: 2463 $GPGSA,A,3,10,27,20,08,21,,,,,,,,3.5,2.6,2.2*3D
age: 1462 $GPGSA,A,3,10,27,20,08,21,,,,,,,,3.5,2.6,2.2*3D

ktuukkan, 2020-04-10:
Hi! Thanks for reporting, I'll make it configurable so you can adjust as needed.

ktuukkan, 2020-04-13:
Hi @aitov, would the above commit resolve the issue for you? The latest snapshot contains the fix now.

aitov, 2020-04-13:
Hi @ktuukkan , I set timeout manually, now working as expected. Thank you.

closed 2020-04-13 by aitov