ktuukkan/marine-api #114: Consider making the SentenceFactory more flexible

ghillert, 2020-06-06:
I am working on parsers for the u-blox PUBX messages [1], [2]. Unfortunately, the *SentenceId* is not enough in this case. The PUBX message types are actually not only identified by *PUBX* but also identified by the secondary *msgId* (Second field of the message):

* $PUBX,00,... -  Lat/Long position data message
* $PUBX,03,... - Satellite status message
* ...

It would be nice if the SentenceFactory would be a tad more flexible and supported some additional/optional identifier logic to support this, admittedly, edge case.

[1] https://gpsd.gitlab.io/gpsd/NMEA.html#_pubx_00_u_blox_latlong_position_data
[2] https://www.u-blox.com/en/docs/UBX-18010854#%5B%7B%22num%22%3A1338%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C59.527%2C672.117%2Cnull%5D

ktuukkan, 2020-06-07:
Hi, is not feasible to resolve this in SentenceFactory, but this looks like similar pattern as in AIS messages, so it could be resolved by adding an additional layer of event listeners on top of PUBX listener/parser. That is, a common handler for all PUBX sentences that checks the first field and then creates a correct parser for the message. You may have a look on [AbstractAISMessageListener](https://github.com/ktuukkan/marine-api/blob/master/src/main/java/net/sf/marineapi/ais/event/AbstractAISMessageListener.java) for more details.

As far as I understood, there is only few different PUBX types, but I don't see problem in adding this as a part of the library nevertheless.

ghillert, 2020-06-09:
Ok - Cool. Thanks for the clarifications. Looks like the AIS route via listeners should work. Working on this this week and might be able to have a PR ready by this weekend to support Pubx00 and Pubx03.

ktuukkan, 2020-06-10:
Ok, that would be nice. I'd suggest creating these in `..sf.marineapi.ublox` package, similarly as with NMEA and AIS. In the `.nmea` package add only the common `UBX` sentence using `P` as the TalkerId. Not sure if a factory class for creating the specific PUBX parsers is needed, but let's see how it looks and works. A simple switch-case or if-else might be enough. I can also refactor/align the details later.

ghillert, 2020-06-16:
Need a few more days. Have a local listener setup that starts to work ;-) but I am hitting some roadblocks - While the AIS Messages are the closest "relative" to Pubx Messages there are some distinct issues. Technically, a Pubx "Message" is not a _Sentence_, yet the parse logic I need is in `SentenceParser`. It is a bit tricky work with composition instead of inheritance.

Another one is the parsing of positional data (For Pubx00). Some of the logic in `PositionParser` could be refactored into something like `PositionParseUtils`. Not sure if I would want _PubxMessages_ to extend _PositionParser_ (SentenceParser).
Maybe a separate ticket, when dealing with Sub-messages (AIS, PUBX), it would be nice if there was better support for manually invoking the message parsing. Right now, it seems the listener approach is the only viable option.

ktuukkan, 2020-06-16:
Ah right.. didn't foresee that earlier, my bad. Admittedly, it is a bit hairy with AIS too but having a look on it as a separate issue is a good idea, if need be. Anyways, I don't see problem if it's done with composition. How about just exposing the required getters from `SentenceParser` and defining a wrapper for each UBX message type with more specific methods?

For example, something like this

```
// low-level UBX handler, basically a regular sentence parser.
public UBXSentence extends Sentence {
  int getUBXType(int);
  String getUBXField(int);
}
UBXParser extends SentenceParser implements UBXSentence {
  int getUBXType() {
    return getIntValue(0);
  }
  String getUBXField(int i) {
    // additional parsing or logic can be done here
    return super.getStringValue(i);
  }
}

// UBXSentence wrapper, i.e. a high-level UBX interface
public UBX01Parser implements UBXMessage01 {
  UBXSentence ubx;
  UBXMessage01(UBXSentence ubx) {
    this.ubx = ubx;
  }
  // here we mostly proxy the calls to corresponding getters/setters
  int getSomeValue() {
     return ubx.getUBXField(SOME_FIELD_IDX);
  }
}
```

Then in the intermediate UBXSentenceListener you'd switch-case the message type and dispatch the corresponding wrapper to the final listener.

ghillert, 2020-06-23:
Cool this works. `UBXMessage03` is working now locally with all data fields being parsed. Still have a to do quite a bit of polishing plus need to add a few test cases. Hopefully just a few more days.

There is a bit of ambiguity, but we can discuss those once the PR is open - e.g. `Position` has an `altitude` property but it is defined as `Gets the position altitude from mean sea level` but the UBX message provides `Altitude above user datum ellipsoid`.

ghillert, 2020-07-06:
Created an initial PR. May still need to do some polishing but it all seems to work. One area I am still frustrated with is the parsing of the satellite ids. This seems to be a bit complicated with PUBX03. Theoretically the satellite ids should allow you to identify the GNSS system - see chapter "1.5.3 Satellite identifiers" - https://www.u-blox.com/en/docs/UBX-18010854 but the `single svId` don't seem to map nicely. Strangely enough `GSV` seem to contain better data in that regard (This NMEA standard is messy ;-).

I hope this PR goes into the right direction. Thanks!

ktuukkan, 2020-07-07:
Ok, thanks for your efforts, I'll have a look in the near days.

closed 2020-10-13 by ktuukkan