ktuukkan/marine-api #116: Add support for Signal ID to GSV Parser

ghillert, 2020-07-14:
NMEA 4.10+ added support for the [GNSS Signal ID](https://www.u-blox.com/en/docs/UBX-18010854#%5B%7B%22num%22%3A1295%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C59.527%2C613.937%2Cnull%5D) to GSV messages.

ghillert, 2020-08-03:
Added PR. I refrained from creating an enum for the signal ids. It looks like devices may send back proprietary values for older protocol versions. Also having the signal field behind a variable-length group, is kind of screwy. I hope the parsing logic makes sense.

<rant>_NMEA_ in general and _GSV_ in particular are really quite messed up. E.g. How do you *know* that you received ALL GSV messages for the cycle? And the signal id function as a key, too (Talker ID + Signal Id) with the added fun that the signal id may be null ...</rant>.

ktuukkan, 2020-08-05:
Yep, in many places it's quite inconsistent and messed up with proprietary conventions. That's what I've always tried to hide and make easier for users in this library, but unfortunately it's not always that easy, given all those oddities.

ghillert, 2020-08-06:
@ktuukkan By the way, I was thinking about the issue of the "reporting cycle". As there is no way to know where the message cycle ("transaction") starts or ends, GPSD does some profiling to determine a complete cycle across GPS devices. Did you ever consider implementing something like this in Marine API? I wonder if that would be a good add-on. It is crazy that this is not covered in the standard.

ktuukkan, 2020-08-07:
SatelliteInfoProvider collects a sequence of GSVs and delivers the data in a single event, but other than that I haven't given it much attention. Now that you meantioned it, the provider should probably be amended with signal ID as well, in any case. And while at it, maybe see if there's also something to improve?