ktuukkan/marine-api #121: Make AIS length validation less restricted

qtran, 2020-09-04:
Hi,
would it be possible to make the AIS length validation less restricted or configurable?
Due to tests with different AIS sources we noticed that sometimes that it is required to increase the maximum length.

Best regards

ktuukkan, 2020-11-17:
Hi @qtran, could you give some examples cases?

RiseOfDeath, 2022-11-21:
Seems that it hardcoded: As example in AISPositionReportParser.java:
`		if (content.length() != 168)
			throw new IllegalArgumentException("Wrong message length");`

ktuukkan, 2022-11-29:
Please provide more context and sample data - or preferrably a pull-request proposing a fix.

ktuukkan, 2024-08-30:
As far as I see, the length is according to spec and arbitrary length would heavily affect payload decoding - e.g. which field(s) should be affected by the added or removed bits? - so I believe the root problem is in your data not respecting the spec. Hence, I don't see any feasible changes to be made, but feel free to propose any improvements you may have.

closed 2024-08-30 by ktuukkan