ktuukkan/marine-api #127: AISPositionReport getSpeedOverGround meaning

Korzi, 2021-01-15:
Hi @ktuukkan 

Could you please help me with understanding `getSpeedOverGround` method in AISPositionReport?
In the javadoc it says:
"return Speed in 1/10 knot steps (0-102.2 knots), 1023 = not available,
             1022 = 102.2 knots or higher."

Does it return 1/10 knots or actual knots? I am not sure what "knot step" is. Is it that "1022" value (value of knots with one decimal place represented by int value) that later is divided by 10 to calculate actual knots? 

Best Regards
Jakub

ktuukkan, 2021-01-19:
Hi @Korzi,

the parser returns actual knots. The original value is delived as integer, so the getter divides it by 10 to get the 1/10 precision (one decimal). Hence the "step" only means the precision of value that's encoded as int (1022, 1023..).

What the javadoc says is directly from [AIS documentation](https://www.navcen.uscg.gov/?pageName=AISMessagesA), so it's a admittely a bit confusing for the marine-api users. It seems the COG has also been documented similarly, so I'll fix both.

Korzi, 2021-01-19:
Thank you for the clarification :) 

Not sure whether I should close the issue or you will close it after fixing docs? Or should I create merge request with this small fix?

ktuukkan, 2021-01-19:
Let's keep it open so I don't forget about it :)

closed 2021-10-09 by ktuukkan