ktuukkan/marine-api #129: event list grows forever if isReady() always returns false for PositionProvider

piemmm, 2021-09-22:
Hi! 

If isReady(), for example in PositionProvider always returns false because no valid GPS signal is ever acquired for a considerable period of time, then the 'events' list will eventually grow big enough to consume all CPU/RAM available to the VM. Typically this takes less than 2 days to happen at the usual 1 second interval.

The events list should be cleared/pruned/expired/etc, or better yet it could be changed to a map, keyed by sentences which are simply replaced when that sentence is updated.

The offending bit of code is here in AbstractProvider.java:
```
public void sentenceRead(SentenceEvent event) {
		events.add(event);
		if (isReady()) {
			if (validate()) {
				T pEvent = createProviderEvent();
				fireProviderEvent(pEvent);
			}
			reset();  <---- never hit if isReady() is false, for example is no signal is aquired for a few days.
		}
	}
```

ktuukkan, 2021-10-05:
Good catch, thanks @piemmm!

ktuukkan, 2021-10-09:
A map would be nice solution to save memory, but it doesn't work if the data spans over multiple sentences of same type, e.g. `GSV` or `RTE,` and the provider must collect all of them before firing an event.

piemmm, 2021-10-12:
Hi!

That looks fine

Worth a mention that if the clock has drifted (and is reset by GPS) then currentTimeMillis can jump forwards or backwards, so you may end up with the possibility of a couple of stale entries depending on how current (or future) providers work (though for this application I doubt it matters that much :-))

closed 2021-12-14 by ktuukkan