ktuukkan/marine-api #130: Android Proguard issue with marine-api

eclectice, 2021-12-06:
Just a reminder that if this API is used in an Android project and the `minifyEnabled true` option is used, the following proguard entry must be added:
```lang-proguard
-keep class net.sf.marineapi.nmea.** { *; }
```

ktuukkan, 2021-12-08:
Hi @eclectice, thanks! I'm not familiar with Android, is there anything that could be done on the library to avoid that?

eclectice, 2021-12-09:
It is highly specialized to Android development, where Android developers can use the Proguard functionality to minimize, optimize, and obfuscate the Java/Kotlin codes...I guess it can be included in the README or Wiki. 

The final Proguard setting that I use right now is:

```lang-proguard
-keep class net.sf.marineapi.** { *; }
-keep interface net.sf.marineapi.** { *; }
-keepattributes MethodParameters
-dontwarn gnu.io.CommPortIdentifier
-dontwarn gnu.io.RXTXPort
-dontwarn gnu.io.SerialPort
```

The marine-api calls are not executed at all if the Proguard configuration is not included in the Android app's `build.gradle` script file.

ktuukkan, 2021-12-14:
Updated README with this info, thanks!

closed 2021-12-14 by ktuukkan