ktuukkan/marine-api #131: AIS position interfaces

alexismanin, 2022-01-20:
I am experimenting AIS message reading, and I am wondering how to access its information in a convenient way.

By looking at AISPositionReport and AISPositionReportB, I see that they contain a lot of similarity that, I think, should be factorized in a common ancestor (maybe in the existing AISPositionInfo interface, but it is not public).

What do you think about it ? Would you like me to do a PR to factorize some fields like COG, heading, position, etc. in a common public ancestor ?

ktuukkan, 2022-01-21:
Yes I see the problem, and I don't see any trouble in declaring AISPositionInfo public. On the NMEA-side we have PositionSentence interface and Position object returned by it. Perhaps the AIS could also provide similar (or even same?) wrapper for position info instead of individual methods? Do you have any additional changes in your mind?

alexismanin, 2022-01-21:
For now, I have no further idea, but I've discovered AIS side of the library recently, so maybe other suggestions will arise as I'll use it.

I'll provide a PR with simple changes as a base for discussion, we can see from here if more should be done.

I'll try to provide the PR in the a few days.

alexismanin, 2022-01-23:
Ok, so after looking at it, I think that : 

1. AISPositionInfo is not the right place to mutualize common properties, because:
    a. Some messages like AISMessage4 depends on it, and they don't provide other properties like the ones in AISPositionReport and AISPositionReportB.
   b. Anyway, in both AISPositionReport and AISPosittionReportB, SOG is defined as double. If we define it in parent AISPositionInfo, the definition would clash with AISMessage09, that defines SOG as int.
2. Because some other public interfaces depends on AISPositionInfo, I think it should be set public anyway.
3. AISPositionReport could just inherit from AISPositionReportB, because the latter only defines fields that can also be found in the former. However, in term of model, a B class message would become parent of "generic class" model, which in turn is parent of the "Class A message".

I think I'll start by exploring point 3, to see where it goes.

ktuukkan, 2022-02-05:
@alexismanin Pull requests merged/closed and deployed a new snapshot.

closed 2022-03-12 by ktuukkan