ktuukkan/marine-api #134: Wrong checksum calculator for NMEA1083 AIS/GPS (AIVDM / G)

YannPara, 2022-08-14:
Hello,
I think it's may have an error in checksum computation for NMEA sentence (or I use wrong your API).

**Describe the bug**
When decode AIS+GPS stream with Marine-API, find >95% of "Invalid data" output (but streams is decode with error<1% on others software).

**To Reproduce**
Sentence sample :  !AIVDM,1,1,,A,133sfv002POVWD0Je4sMA:a@00S?,0*3F
Compute CRC with Marine-api : 7A
Compute CRC with online decoder (links at bottom) & other software : 3F

Sentence sample : !AIVDM,1,1,,A,13IWB67P00wdI9pKF>atlww>2@G4,0*4F
Compute CRC with Marine-api : 0F
Compute CRC with online decoder (links at bottom) & other software : 4F

For test I use directly net.sf.marineapi.nmea.sentence.SentenceValidator  "public static boolean isValid(String nmea)" 

Sometime compute is good, like in this sample :
!AIVDM,1,1,,B,1CI7wSg000woG1rK3DMh0q3D00RV,0*6D

**Configuration** :
OS: WIN11
JDK OpenJDK 17.0.3.v202205015-1416
Marine-API version 0.11.0

**Additional context**
[Online checkSum validator N°1](https://alekunderwater.github.io/nmea0183_checksum_calculator.html)
[Online checkSum validator N°2](http://www.hhhh.org/wiml/proj/nmeaxor.html)
[Online checkSum validator N°3](https://www.scadacore.com/tools/programming-calculators/online-checksum-calculator/)

** Fast fix before API-update **
Remove all characters after * before parsing, so validator skip checksum control, but real invalid data can pass.

** Better fix before API-update **
Remove all characters after * before parsing, outside API checksum validor, if OK, pass data (with no CRC) to API.

Best regards,

ktuukkan, 2022-08-16:
Hi @YannPara, thanks for reporting!

I added some unit tests for the samples you provided, but I wasn't able to reproduce the mentioned results. All tests pass checksums matching and the sentence validation passes too (see commit da775b2).

That said, I didn't run the tests against v0.11 yet, so this might have been already fixed in the latest snapshot.

ktuukkan, 2023-02-26:
@YannPara Checked also against v0.11 tag, but still wasn't able to reproduce the issue.

closed 2024-08-30 by ktuukkan