ktuukkan/marine-api #137: Wrong message handle

RiseOfDeath, 2022-11-25:
Hi all, can you advice how to handle wrong messages (f.e. java.lang.IllegalStateException: java.lang.IllegalArgumentException: Wrong message length).

Exeption listener, which I have setted for reader, not called for such type of exeptions.

ktuukkan, 2022-11-29:
Duplicate of #121

closed 2022-11-29 by ktuukkan

RiseOfDeath, 2022-12-08:
It's not duplicate. 
121 about message lenght restrictions.

This one about exeption handling (exeption with length just example of exeption which not handled by listenep provided by setExceptionListener()

ktuukkan, 2022-12-08:
@RiseOfDeath The exception listener is not meant to receive validation or parsing errors, only exceptions related to data streams reading. This is to enable you to better recover at runtime of sudden disconnect or such. It may not be expressed clearly enough in the javadocs, but if you look at the ExceptionListener, it says:

> A listener callback interface for listening to Exceptions **in DataReaders**.

ktuukkan, 2022-12-08:
@RiseOfDeath The exception listener is not meant to receive validation or parsing errors, only exceptions related to data streams reading. This is to enable you to better recover at runtime of sudden disconnect or such. It may not be expressed clearly enough in the javadocs, but if you look at the ExceptionListener, it says:

> A listener callback interface for listening to Exceptions **in DataReaders**.
> ...
> Invoked by {@link SentenceReader} when error has occured while reading the data source.
> ...
> Exception that was thrown while reading data.

EDIT: On the other hand, if you look at the `AbstractDataReader#run()`, it looks like the exceptions from `SentenceFactory` could indeed end up to `ExceptionListener`. However, the normal NMEA sentences, including the AIS `AIVDM` and `AIVDO`, are validated before actual parsing and therefore it is quite unprobable that such exceptions would ever be thrown. Moreover, as we are talking about the AIS layer on top of regular NMEA, the exception you are seeing happens on the next level in `AbstractAISMessageListener#sentenceRead()`, meaning the data reading and NMEA validation has already passed. But for some reason, your data does not pass the AIS message validation.

ktuukkan, 2022-12-08:
@RiseOfDeath The exception listener is not meant to receive validation or parsing errors, only exceptions related to data streams reading. This is to enable you to better recover at runtime of sudden disconnect or such. It may not be expressed clearly enough in the javadocs, but if you look at the ExceptionListener, it says:

> A listener callback interface for listening to Exceptions **in DataReaders**.
> ...
> Invoked by {@link SentenceReader} when error has occured while reading the data source.
> ...
> Exception that was thrown while reading data.

On the other hand, if you look at the `AbstractDataReader#run()`, it looks like the exceptions from `SentenceFactory` could indeed end up to `ExceptionListener`. However, the normal NMEA sentences, including the AIS `AIVDM` and `AIVDO`, are validated before actual parsing and therefore it is quite unprobable that such exceptions would ever be thrown. Moreover, as we are talking about the AIS layer on top of regular NMEA, the exception you are seeing happens on the next level in `AbstractAISMessageListener#sentenceRead()`, meaning the data reading and NMEA validation has already passed. But for some reason, your data does not pass the AIS message validation.