ktuukkan/marine-api #138: Position accuracy always false

RiseOfDeath, 2023-02-08:
Position accuracy always have "false" value.
Example message with this problem.

> 
!AIVDM,1,1,,A,15S0t`001TlGn>TNurwroHgD05;H,0*21

ktuukkan, 2023-02-10:
Hi, added a test for the given sentence, but it seems to pass as "true". Any other sentences you are having trouble with?

ktuukkan, 2023-02-11:
@RiseOfDeath, investigated further today and it passes also when parsed with `AISMessageFactory`, which identifies it as type 1 / position report A. I also get the same results with the [online decoder](http://ais.tbsalling.dk/).

The only way I could produce a failure is by trying to parse the message as type 2 / report B, but it requires access to the package private constructor of `AISMessage02Parser`, which should not happen in normal use.

ktuukkan, 2023-02-11:
@RiseOfDeath, investigated further and it passes also when parsed with AISMessageFactory, which identifies it as type 1 / position report A. I also get the same results with the [online decoder](http://ais.tbsalling.dk/).

The only way I could produce a failure is by trying to parse the message as type 2 / report B, but it requires access to the package private constructor of AISMessage02Parser, which should not happen in normal use.

ktuukkan, 2023-02-11:
@RiseOfDeath, investigated further and it passes also when parsed with `AISMessageFactory`, which identifies it as type 1 / position report A. I also get the same results with the [online decoder](http://ais.tbsalling.dk/).

The only way I could produce a failure is by trying to parse the message as type 2 / report B, but it requires access to the package private constructor of `AISMessage02Parser` which should never happen in normal use.

ktuukkan, 2023-02-26:
@RiseOfDeath I've now confirmed this is a bug in v0.11.0 but the latest snapshot isn't affected. I should also probably publish v0.12 as it's been waiting for quite some time already.

Test result when running against the tag of v0.11.0:
```
[INFO] Running net.sf.marineapi.ais.parser.AISMessageFactoryTest
[ERROR] Tests run: 4, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 0.004 s <<< FAILURE! - in net.sf.marineapi.ais.parser.AISMessageFactoryTest
[ERROR] net.sf.marineapi.ais.parser.AISMessageFactoryTest.testCreateIssue138  Time elapsed: 0 s  <<< FAILURE!
java.lang.AssertionError
        at net.sf.marineapi.ais.parser.AISMessageFactoryTest.testCreateIssue138(AISMessageFactoryTest.java:42)
```

RiseOfDeath, 2023-03-09:
I can prove that in v0.12.0 I have no such problem, so seems that bug fixed.

closed 2023-03-09 by RiseOfDeath