ktuukkan/marine-api #149: ZDA parse fails to parse sentence with the local hour and minutes not set

minghia, 2024-09-01:
Our ZDA generator that we decode using marine-api, the getTime method fails with an exception and so any attempt to get the time represented by a sentence with the local hour and minute offset fields left empty fails.

ktuukkan, 2024-09-23:
@minghia The parser indeed requires both time and offset to be present, based on the assumption that devices will report offset because otherwise it is rather unambiguous. But I guess marine-api could default to UTC-0 when not specified, although it might not be correct.

minghia, 2024-09-24:
The NMEA 0183 document is vague in what fields are necessary. The general comment about leaving fields blank if they are not provided doesn't help either.

I don't know about commercial ships, but navy ships usually work in UTC time so I can see that the UTC offset fields could be empty. I'm of the opinion that if field is not present then the parsing shouldn't fail, but that is just my opinion.