ktuukkan/marine-api #150: Throw exceptions on optional fields

FraRo96, 2024-11-08:
Why when a field is not present it throws an exception instead of just assigning null? I looked for mandatory vs optional fields in nmea 0183, looks like there aren't explicit mandatory rules for any field, so why throwing an exception that ruins the whole multi-threaded execution of an app.

ktuukkan, 2024-11-26:
Hi @FraRo96, this is a design decision that was made already somewhere around 2005 and since there hasn't been pressure to change it, that's how it has remained. Surely you are not the first one to ask about this, but nevertheless among the very few. In general, using try-catch shouldn't be too tedious in contrast to the fact that you'll most likely need to if-else the null values too. I believe this is also somewhat a matter of taste. With individual values/nulls checking you'd need to repeat the check for every value you are trying to get. So I found it more convenient to wrap sentence handling in a single try-catch and if any of the values that my app requires is missing, then I end up in the catch block.

> throwing an exception that ruins the whole multi-threaded execution of an app

For me to understand better, could you elaborate how it ruins the threading / execution for you?

FraRo96, 2024-11-26:
I really appreciate your library and your work, nonetheless it's a bad design choice that could easily be fixed though. Maybe I can open a PR if I have time.

I can quote any source about the previous statement. It's stated so clearly [here](https://isocpp.org/wiki/faq/exceptions#why-not-exceptions): _do not use exceptions for control flow. throw is not simply an alternative way of returning a value from a function_. It's cpp docs but it's recommended in every language afaik. Regarding specifically Java, and to answer this

> For me to understand better, could you elaborate how it ruins the threading / execution for you?

ktuukkan, 2024-12-10:
While I don't want to start arguing over rather elementary stuff, I do must point out that the document you linked also gives good advice on using exceptions and I see many of them supporting the chosen approach. For example in [here](https://isocpp.org/wiki/faq/exceptions#why-exceptions) and [here](https://isocpp.org/wiki/faq/exceptions#exceptions-avoid-two-return-types). The section you referred to, also says (emphasis by me):

> Use throw only to signal an error (which means ***specifically that the function couldn’t do what it advertised***, and establish its postconditions).

> In particular, do not use exceptions for control flow. throw is not simply an alternative way of returning a value from a function ***(similar to return)***

In my honest opinion, marine-api is not throwing to "return values" and especially not to control program flow. It is simply a signal of runtime error and enables error handling as it is conventional in Java.

But yeah, let's see this from more practical point of view. The `SentenceParser` has been defined using native data types in order to avoid the overhead of creating corresponding data type objects (Double, Integer etc). Efficiency is also the reason why all sentence fields are not parsed immediately after reading the data, but mostly individually and on-demand. Java autoboxing/unboxing can handle converting non-String data types if we return `null` from `SentenceParser#getStringValue(int)`, but in all other non-String getters there is an additional layer of parsing/conversion. For example, we convert String into a double or boolean. If this fails for some reason, be it due to malformed or missing value, an exception will be thrown by Java standard lib methods (eg. `NumberFormatException`). So there will be exceptions and if we also start returning nulls, you need to take care of both worlds. There will also be some additional things you need to consider as the user of marine-api, for example assigning nulls to native type variables and equality comparisons:

```
double alt1 = gga1.getAltitude(); // throws NPE if altitude is null
double alt2 = gga2.getAltitude();
if (alt1 == alt2) {
    // we will never get here as we are comparing object refs
}
```

Hence, I've chosen not to return nulls. But if you still feel it's better and makes everyones life happier, feel free to refactor and open a PR and we'll see it from there.

ktuukkan, 2024-12-10:
While I don't want to start arguing over it, I do must point out that the document you linked also gives good advice on using exceptions and I see many of them supporting the chosen approach. For example in [here](https://isocpp.org/wiki/faq/exceptions#why-exceptions) and [here](https://isocpp.org/wiki/faq/exceptions#exceptions-avoid-two-return-types). The section you referred to, also says (emphasis by me):

> Use throw only to signal an error (which means ***specifically that the function couldn’t do what it advertised***, and establish its postconditions).

> In particular, do not use exceptions for control flow. throw is not simply an alternative way of returning a value from a function ***(similar to return)***

In my honest opinion, marine-api is not throwing to "return values" and especially not to control program flow. It is simply a signal of runtime error and enables error handling as it is conventional in Java.

But yeah, as for some more background on the decision, let's see this from more practical point of view. The `SentenceParser` has been defined using native data types in order to avoid the overhead of creating corresponding data type objects (Double, Integer etc). Efficiency is also the reason why all sentence fields are not parsed immediately after reading the data, but mostly individually and on-demand. Java autoboxing/unboxing can handle converting non-String data types if we return `null` from `SentenceParser#getStringValue(int)`, but in all other non-String getters there is an additional layer of parsing/conversion. For example, we convert String into a double or boolean. If this fails for some reason, be it due to malformed or missing value, an exception will be thrown by Java standard lib methods (eg. `NumberFormatException`). So there will be exceptions and if we also start returning nulls, you need to take care of both worlds. There will also be some additional things you need to consider as the user of marine-api, for example assigning nulls to native type variables and equality comparisons:

```
double alt1 = gga1.getAltitude(); // throws NPE if altitude is null
double alt2 = gga2.getAltitude();
if (alt1 == alt2) {
    // we will never get here as we are comparing object refs
}
```

Hence, I've chosen not to return nulls. But if you still feel it's better and makes everyones life happier, feel free to refactor and open a PR and we'll see it from there.

ktuukkan, 2024-12-10:
While I don't want to start arguing over it, I do must point out that the document you linked also gives good advice on using exceptions and I see many of them supporting the chosen approach. For example in [here](https://isocpp.org/wiki/faq/exceptions#why-exceptions) and [here](https://isocpp.org/wiki/faq/exceptions#exceptions-avoid-two-return-types). The section you referred to, also says (emphasis by me):

> Use throw only to signal an error (which means ***specifically that the function couldn’t do what it advertised***, and establish its postconditions).

> In particular, do not use exceptions for control flow. throw is not simply an alternative way of returning a value from a function ***(similar to return)***

In my honest opinion, marine-api is not throwing to "return values" and especially not to control program flow. It is simply a signal of runtime error and enables error handling as it is conventional in Java.

But yeah, as for some more background on the decision, let's see this from more practical point of view. The `SentenceParser` has been defined using native data types in order to avoid the overhead of creating corresponding data type objects (Double, Integer etc). Efficiency is also the reason why all sentence fields are not parsed immediately after reading the data, but mostly individually and on-demand. Java autoboxing/unboxing can handle converting non-String data types if we return `null` from `SentenceParser#getStringValue(int)`, but in all other non-String getters there is an additional layer of parsing/conversion. For example, we convert String into a double or boolean. If this fails for some reason, be it due to malformed or missing value, an exception will be thrown by Java standard lib methods (eg. `NumberFormatException`). So there will be exceptions and if we also start returning nulls, you need to take care of both worlds. And because of autoboxing, there will also be some additional things you need to remember as the user of marine-api, for example the null assignment to native type variables and caveats with equality comparisons (object refs vs. values). Hence, I've chosen not to return nulls. But if you feel it's better and benefits everyone, feel free to refactor and open a PR and we'll see it from there.

FraRo96, 2024-12-10:
Take constructive criticism however you prefer, however it’s a section to point out issues.
Let me rephrase my first objection, as as it firstly appeared so rather elementary arguing, my bad. You consider a runtime error the absence of value, while the protocol accepts such absence (please prove me wrong on this if I am). Therefore, you’re declaring that your library is not designed for apps using gnss receivers that, for a bunch of reasons like temporary bad or missing gnss signal, sometimes leave empty some fields of a sentence. This means to throw exceptions and to cause performance issues as stated in the benchmark I previously linked. I think it’s pretty practical use case, so prove me wrong on this.

The rest of the answer if I interpret well is you saying it’s a pain to do null checks and not using primitive types, but actually it’s not, that’s why e.g. in Kotlin the primitive types are only in the form of wrapper nullable classes.

FraRo96, 2024-12-10:
I’m quite busy lately because I’m working on my first open source library during spare time, but I’ll consider your PR call sooner or later, have a good day.

closed 2026-09-13 by ktuukkan